hencou/esphome_components #1: Pairing does'nt work? is this a WiP project?

jetflux, 2021-12-10:
Hi, really excellent work!!!!. is this a WiP?, not ready?.
Because I cannot get it to work. It connects to API in homeassistant, but pariing doesn't work. 
I'm trying to use it with a FUT092 RGB-CCT. 
I've use/used it with a wemos d1 mini. with sidoh/esp8266_milight_hub it works great, but this with esphome is more what I want/look for.

From serial "Opening settings file failed" but then it connects, I tried different boards. erase_flash etc etc...

Pairing in homeassistant doesn't work. it connects to board API when I press pair button but nothing happens, I had a look with tcpdump.

hencou, 2021-12-10:
Hi, its in production on 20+ devices, since a month and all working stable. Pairing / unpairing also works. Can you post an example of your configured yaml?

hencou, 2021-12-10:
So you paired the FUT092 RGB-CCT bulb already with sidoh/esp8266_milight_hub, right? So the bulb should already respond to commands send to 0xFFF1 and group_id 1. Does it respond on bulb commands like on/off brigness etc?

jetflux, 2021-12-11:
Yes it's paired try it with unpair first also. no respons on/off. unpair/pair...

I get some compile errors also maybe related but only a warning? deprecated should not be any problem?

src/milight/Settings/Settings.cpp: In static member function 'static void Settings::load(Settings&)':
src/milight/Settings/Settings.cpp:211:7: warning: 'SPIFFS' is deprecated (declared at /home/jetflux/.platformio/packages/framework-arduinoespressif8266@3.20704.0/cores/esp8266/FS.h:269): SPIFFS has been deprecated. Please consider moving to LittleFS or other filesystems. [-Wdeprecated-declarations]
   if (SPIFFS.exists(SETTINGS_FILE)) {

and also this dont understand?

.piolibdeps/workdesk/RGBConverter/RGBConverter.cpp: In member function 'void RGBConverter::rgbToHsl(byte, byte, byte, double*)':
.piolibdeps/workdesk/RGBConverter/RGBConverter.cpp:45:15: warning: 'h' may be used uninitialized in this function [-Wmaybe-uninitialized]
         h /= 6;
               ^
.piolibdeps/workdesk/RGBConverter/RGBConverter.cpp: In member function 'void RGBConverter::rgbToHsv(byte, byte, byte, double*)':
.piolibdeps/workdesk/RGBConverter/RGBConverter.cpp:112:15: warning: 'h' may be used uninitialized in this function [-Wmaybe-uninitialized]
         h /= 6;
               ^
.piolibdeps/workdesk/RGBConverter/RGBConverter.cpp: In member function 'void RGBConverter::hsvToRgb(double, double, double, byte*)':
.piolibdeps/workdesk/RGBConverter/RGBConverter.cpp:151:16: warning: 'b' may be used uninitialized in this function [-Wmaybe-uninitialized]
     rgb[2] = b * 255;
                ^
.piolibdeps/workdesk/RGBConverter/RGBConverter.cpp:150:16: warning: 'g' may be used uninitialized in this function [-Wmaybe-uninitialized]
     rgb[1] = g * 255;
                ^
.piolibdeps/workdesk/RGBConverter/RGBConverter.cpp:149:16: warning: 'r' may be used uninitialized in this function [-Wmaybe-uninitialized]
     rgb[0] = r * 255;
                ^

hencou, 2021-12-11:
Hi Jetflux,

The compile errors are ony warnings in external libraries. The module should work fine with this.

But when nothing responds on your commands: maybe a problem in the connection between the Wemos D1 and NRF24 module? With the default settings, CE pin in this project is connected to GPIO16. In the orignal espMH project it is default connected to GPIO4. This is because I am using pins D1-4 for a wall switch to turn group 1-4 on or off.

jetflux, 2021-12-11:
Hi, Excellent. that was it :) I just switched the CE pin to GPIO16 and now it works perfect!!!! 

Thank you very much!, for this. sorry for bothering you with this. thank you again!

hencou, 2021-12-11:
You're welcome, have fun with it! I have slightly modified the documentation so thats clear for everybody in the future.

closed 2021-12-12 by hencou