hencou/esphome_components #2: Compiling for ESP32

ashp8i, 2021-12-23:
is this code ready/doable for a ESP32?

hencou, 2021-12-23:
It is only for ESP8266

ashp8i, 2021-12-23:
please close issue if this is a huge effort to support esp32

hencou, 2021-12-23:
Hi,

Seems possible. see example here: https://github.com/hencou/esphome_components/blob/main/example_milight_esp32.yaml
I have no ESP32 to test with, but compiling fine.

ashp8i, 2021-12-23:
Thanks for trying to test with ESP32 setup, I do have both modules, however have issues with getting the code to compile in ESPHome 2021.12.1.

I get the following below: -

`
src/esphome/components/mi/mi.cpp: In member function 'virtual void esphome::mi::Mi::setup()':
src/esphome/components/mi/mi.cpp:258:7: error: 'SPIFFS' was not declared in this scope
       SPIFFS.begin();
       ^
*** [/data/mitest/.pioenvs/mitest/src/esphome/components/mi/mi.cpp.o] Error 1
========================= [FAILED] Took 49.44 seconds =========================
`

hencou, 2021-12-23:
Maybe you are using an old version of the libraries?
Please clean the directory "/config/esphome/.esphome/external_components", then new versions of the files are retrieved from Github. And please replace the "milight" folder in the root of your esphome folder with the latest version from Github.

ashp8i, 2021-12-24:
no sure what I'm doing wrong keep getting: -

`Compiling /data/mitest/.pioenvs/mitest/src/MyMiLight.cpp.o
In file included from src/esphome.h:19:0,
                 from src/MyMiLight.cpp:1:
src/esphome/components/json/json_util.h:27:46: error: 'ArduinoJson::Internals' has not been declared
 class VectorJsonBuffer : public ArduinoJson::Internals::JsonBufferBase<VectorJsonBuffer> {
                                              ^
src/esphome/components/json/json_util.h:27:57: error: expected '{' before 'JsonBufferBase'
 class VectorJsonBuffer : public ArduinoJson::Internals::JsonBufferBase<VectorJsonBuffer> {
                                                         ^
src/esphome/components/json/json_util.h:27:71: error: expected initializer before '<' token
 class VectorJsonBuffer : public ArduinoJson::Internals::JsonBufferBase<VectorJsonBuffer> {
                                                                       ^
src/MyMiLight.cpp:484:1: error: expected '}' at end of input
 }
 ^
src/MyMiLight.cpp:484:1: error: expected '}' at end of input
*** [/data/mitest/.pioenvs/mitest/src/MyMiLight.cpp.o] Error 1
========================== [FAILED] Took 7.16 seconds ==========================
`

Kind Regards,
Ash

ashp8i, 2021-12-24:
I think I've caught it: -

`|-- <ArduinoJson> 6.18.5
|-- <WiFi> 1.0
|-- <Update> 1.0
|-- <ESPAsyncWebServer-esphome> 2.1.0
|   |-- <AsyncTCP-esphome> 1.2.2
|-- <DNSServer> 1.1.0
|-- <ArduinoJson-esphomelib> 5.13.3
`

ArduinoJson-esphomelib is v5.13.3

where as ArduinoJson is v6.18.5.

This causes breaking changes, I can see your code is using the newer ArduinoJson (bblanchon/ArduinoJson) which is v6.18.5 28 sep 2021

ashp8i, 2021-12-24:
zapped esphome, restoring older version esphome 2021.12.0 to test

ashp8i, 2021-12-24:
> Hi,
> 
> Seems possible. see example here: https://github.com/hencou/esphome_components/blob/main/example_milight_esp32.yaml I have no ESP32 to test with, but compiling fine.

This works therefore I can compile, now testing

closed 2021-12-24 by ashp8i