hencou/esphome_components #3: Select Pins on ESP32

ashp8i, 2021-12-24:
I've managed to get this going on an esp8266 with no webserver as that conflicts library version of ArduinoJson used by webserver, how do I specify which pins on an esp32?

hencou, 2021-12-24:
Hi Ash,

The webserver part is not needed anymore. Most of the ssettings are fixed in this file: https://github.com/hencou/esphome_components/blob/main/milight/Settings/Settings.h. Eventually it is possible to make them as variables in ESPHome, but for me this approach was enough.

hencou, 2021-12-29:
Hi,

Created an updated version, now it is possible to configure the pins and so on in config yaml:

> mi:
  ce_pin: D0
  csn_pin: D8
  radio_interface_type: nrf24
  packet_repeats: 50
  listen_repeats: 20
  state_flush_interval: 5000
  packet_repeat_throttle_threshold: 200
  packet_repeat_throttle_sensitivity: 10
  packet_repeat_minimum: 3
  enable_automatic_mode_switching: false
  rf24_power_level: MAX
  rf24_listen_channel: LOW
  packet_repeats_per_loop: 10

closed 2021-12-29 by hencou