hencou/esphome_components #4: Recommendation: mihome folder to separate repo as PlatformIO Library

shbatm, 2022-01-02:
Wanted to recommend that you split the `mihome` folder into a separate repository and structure it like a PlatformIO library.

Reason: this allows you to include the library from GitHub in your ESPHome config instead of needing to copy it to the directory manually and update it every time changes are made to the repo. 

I've already created a test repo and tested this method using the following snippet:

```yaml
esphome:
  name: $device_name
  platform: ESP8266
  board: d1_mini
  libraries:
    - milight=https://github.com/shbatm/esphome-milight#main
```

The only issue I found was that I had to directly incorporate the color conversion functions instead of referencing the `esphome.h` functions -- the way PlatformIO compiles the library, using this as is would mean adding the full `esphome` library as a dependency for only these two functions.

I made the repo for the library here: https://github.com/shbatm/esphome-milight; I'd be happy to transfer ownership to you if you want to use this method.

Thanks again for all the work!

hencou, 2022-01-02:
Hi, I am happy with your suggestion! Was searching on a solution like this.
I will have a look for an alternative library for that RGB<>HSV functions,
moved that from RGB converter library to ESP home function calls. But in
this approach maybe it is better to use ColorConverter library for example,
then we don't need the whole esphome included.

shbatm, 2022-01-02:
Those functions are pretty small and not likely to change anytime soon--you could add another library or just leave the functions copied in like I did here: https://github.com/shbatm/esphome-milight/blob/01dcd7ca6373b140808cd08fdac6e52e39058ae2/src/milight/MiLightState/GroupState.cpp#L992

I sent you a request to transfer the repository to your ownership. You can take over from there--you'll need to change the path in the `library.json` file back you your username.

shbatm, 2022-01-02:
Also -- just FYI, I used `git filter-repo` to create the other repo from this one, so you should still have all the applicable commit history.

hencou, 2022-01-02:
Thank you very much! Tested here, everything is working fine with your modifications.

closed 2022-01-02 by hencou