hencou/esphome_components #7: This message is dropped by ESPHome when installing the sample code.

winczead, 2022-01-09:
INFO Reading configuration /config/esphome/mi-test.yaml...
INFO Generating C++ source...
INFO Core config or version changed, cleaning build files...
INFO Compiling app...
Processing mitest1 (board: d1_mini; framework: arduino; platform: platformio/espressif8266 @ 2.6.3)
--------------------------------------------------------------------------------
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
Library Manager: Installing git+https://github.com/hencou/esphome-milight#main
git version 2.30.2
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-uvkecpyl'...
Library Manager: milight @ 0.0.2+sha.43e140a has been installed!
Library Manager: Installing dependencies...
Library Manager: Installing SPI @ *
Library Manager: Installing nrf24/RF24 @ *

Unpacking  [------------------------------------]    0%
...
Unpacking  [####################################]  100%
Library Manager: RF24 @ 1.4.2 has been installed!
Library Manager: Installing PathVariableHandlers @ *

Unpacking  [------------------------------------]    0%
...
Unpacking  [####################################]  100%
Library Manager: PathVariableHandlers @ 2.0.0 has been installed!
Library Manager: Installing ListLib @ *

Unpacking  [------------------------------------]    0%
...
Unpacking  [####################################]  100%
Library Manager: ListLib @ 1.0.0 has been installed!
Library Manager: Installing bblanchon/ArduinoJson @ *

Unpacking  [------------------------------------]    0%
...
Unpacking  [####################################]  100%
Library Manager: ArduinoJson @ 6.19.0 has been installed!
Library Manager: RF24 @ 1.4.2 is already installed
Library Manager: PathVariableHandlers @ 2.0.0 is already installed
Library Manager: Installing git+https://github.com/luisllamasbinaburo/Arduino-List
git version 2.30.2
Cloning into '/data/cache/platformio/cache/tmp/pkg-installing-tppzaa4w'...
Library Manager: ListLib @ 1.0.0+sha.a250c6e has been installed!
Library Manager: ArduinoJson @ 6.19.0 is already installed
Dependency Graph
|-- <milight> 0.0.2+sha.43e140a
|   |-- <SPI> 1.0
|   |-- <RF24> 1.4.2
|   |-- <PathVariableHandlers> 2.0.0
|   |-- <ListLib> 1.0.0
|   |-- <ArduinoJson> 6.19.0
|-- <ESP8266WiFi> 1.0
|-- <SPI> 1.0
|-- <RF24> 1.4.2
|-- <PathVariableHandlers> 2.0.0
|-- <ListLib> 1.0.0+sha.a250c6e
|-- <ArduinoJson> 6.19.0
|-- <ESP8266mDNS> 1.2
Compiling /data/mitest1/.pioenvs/mitest1/src/esphome/components/api/api_connection.cpp.o
Compiling /data/mitest1/.pioenvs/mitest1/src/esphome/components/api/api_frame_helper.cpp.o
Compiling /data/mitest1/.pioenvs/mitest1/src/esphome/components/api/api_pb2.cpp.o
Compiling /data/mitest1/.pioenvs/mitest1/src/esphome/components/api/api_pb2_service.cpp.o
Compiling /data/mitest1/.pioenvs/mitest1/src/esphome/components/api/api_server.cpp.o
Compiling /data/mitest1/.pioenvs/mitest1/src/esphome/components/api/list_entities.cpp.o
Compiling /data/mitest1/.pioenvs/mitest1/src/esphome/components/api/proto.cpp.o
Compiling /data/mitest1/.pioenvs/mitest1/src/esphome/components/api/subscribe_state.cpp.o
Compiling /data/mitest1/.pioenvs/mitest1/src/esphome/components/api/user_services.cpp.o
Compiling /data/mitest1/.pioenvs/mitest1/src/esphome/components/api/util.cpp.o
Compiling /data/mitest1/.pioenvs/mitest1/src/esphome/components/binary_sensor/automation.cpp.o
Compiling /data/mitest1/.pioenvs/mitest1/src/esphome/components/binary_sensor/binary_sensor.cpp.o
[... 51 lines cut ...]
Compiling /data/mitest1/.pioenvs/mitest1/liba94/milight/milight/MiLight/CctPacketFormatter.cpp.o
Archiving /data/mitest1/.pioenvs/mitest1/libf5a/libSPI.a
Compiling /data/mitest1/.pioenvs/mitest1/liba94/milight/milight/MiLight/FUT020PacketFormatter.cpp.o
Archiving /data/mitest1/.pioenvs/mitest1/lib1cc/libRF24.a
Compiling /data/mitest1/.pioenvs/mitest1/liba94/milight/milight/MiLight/FUT02xPacketFormatter.cpp.o
Compiling /data/mitest1/.pioenvs/mitest1/liba94/milight/milight/MiLight/FUT089PacketFormatter.cpp.o
Compiling /data/mitest1/.pioenvs/mitest1/liba94/milight/milight/MiLight/FUT091PacketFormatter.cpp.o
Compiling /data/mitest1/.pioenvs/mitest1/liba94/milight/milight/MiLight/MiLightClient.cpp.o
Compiling /data/mitest1/.pioenvs/mitest1/liba94/milight/milight/MiLight/MiLightRemoteConfig.cpp.o
Compiling /data/mitest1/.pioenvs/mitest1/liba94/milight/milight/MiLight/PacketFormatter.cpp.o
/data/mitest1/.piolibdeps/mitest1/milight/src/milight/MiLight/MiLightClient.cpp: In member function 'void MiLightClient::update(ArduinoJson::JsonObject)':
/data/mitest1/.piolibdeps/mitest1/milight/src/milight/MiLight/MiLightClient.cpp:350:48: error: 'ArduinoJson::JsonVariant' has no member named 'isUndefined'
   const bool isBrightnessDefined = !brightness.isUndefined() || !level.isUndefined();
                                                ^
/data/mitest1/.piolibdeps/mitest1/milight/src/milight/MiLight/MiLightClient.cpp:350:72: error: 'ArduinoJson::JsonVariant' has no member named 'isUndefined'
   const bool isBrightnessDefined = !brightness.isUndefined() || !level.isUndefined();
                                                                        ^
/data/mitest1/.piolibdeps/mitest1/milight/src/milight/MiLight/MiLightClient.cpp: In member function 'uint8_t MiLightClient::parseStatus(ArduinoJson::JsonVariant)':
/data/mitest1/.piolibdeps/mitest1/milight/src/milight/MiLight/MiLightClient.cpp:457:11: error: 'ArduinoJson::JsonVariant' has no member named 'isUndefined'
   if (val.isUndefined()) {
           ^
Compiling /data/mitest1/.pioenvs/mitest1/liba94/milight/milight/MiLight/PacketQueue.cpp.o
Compiling /data/mitest1/.pioenvs/mitest1/liba94/milight/milight/MiLight/PacketSender.cpp.o
*** [/data/mitest1/.pioenvs/mitest1/liba94/milight/milight/MiLight/MiLightClient.cpp.o] Error 1
========================= [FAILED] Took 26.87 seconds =========================

hencou, 2022-01-09:
Hi winczread,

I changed the code in https://github.com/hencou/esphome-milight/blob/main/src/milight/MiLight/MiLightClient.cpp from .isUndefined to .isUnbound. When you clean the build files, it should donwload the latest version from Github and compiling fine.

winczead, 2022-01-09:
It works! You are a hero :)

closed 2022-01-09 by winczead