hencou/esphome_components #8: esp32 compatible?

jetflux, 2022-01-26:
Hi, excuse me for asking here, but can I use this with esp32? or is it only esp8266.

hencou, 2022-01-26:
Hi Jetflux,

I've never tried this at myself, but I expect that this will work. Do you experience a specific error?

shbatm, 2022-01-26:
Yes, it is working for me on a ESP32 D1 Mini board.

hencou, 2022-01-26:
Great! I will close the ticket then.

closed 2022-01-26 by hencou

jetflux, 2022-01-27:
Hi!, i tried compile it and it works. but get small error:
``
 .platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c: In function 'spiTransferBytesNL':
``

 ``
.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:922:39: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_out8 = &result[c_longs-1];
``

``
.platformio/packages/framework-arduinoespressif32/cores/esp32/esp32-hal-spi.c:923:40: warning: initialization from incompatible pointer type [-Wincompatible-pointer-types]
                 uint8_t * last_data8 = &last_data;
                                        ^
``

maybe nothing to worry about? , I will try to upload and wire it to a esp32 this weekend.

hencou, 2022-01-27:
Most of times you can ignore these kind of warnings. Its a warning coming from the SPI ESP32 library it seems.

poudenes, 2022-12-07:
I have a AZ Delivery D1 R32 board. But what I try on pins its not working for me at all.
With a Wemos D1 it is working, but changed because of bigger ram. Now I can add all my 36 bulbs in config.

Can someone help me with the right pinout to let this work?