hencou/esphome_components #10: Feature Request: Add Trigger for Code Received

shbatm, 2022-04-22:
Have a feature request for you--I could probably clobber something together but my C++ is terrible. 

Add a `on_code_received` trigger & callback to allow automations for remotes not bound to a bulb, similar to the RFBridge Component:

https://esphome.io/api/classesphome_1_1rf__bridge_1_1_r_f_bridge_received_code_trigger.html
https://esphome.io/api/classesphome_1_1rf__bridge_1_1_r_f_bridge_component.html#a8d44e0233fe8f3e405563d6ee5dbf849

Reason: this would allow for extra remotes to be used and events sent back to home assistant using ESPHome Automations.

For example (using [RF Bridge](https://esphome.io/components/rf_bridge.html) as a reference):

```yaml
rf_bridge:
  on_code_received:
    then:
      - homeassistant.event:
          event: esphome.milight_code_received
          data:
            key: !lambda 'return format_hex(data.key);'
            id: !lambda 'return format_hex(data.id);'
            command: !lambda 'return format_hex(data.command);'
            argument: !lambda 'return format_hex(data.argument);'
            group: !lambda 'return format_hex(data.group);'
```

hencou, 2022-04-23:
Hi shbatm,

Seems possible. See the modified yaml example: [https://github.com/hencou/esphome_components/blob/main/example_milight.yaml](url)

closed 2022-04-28 by hencou

shbatm, 2022-04-28:
Awesome, thanks! I didn't see any issues with the code, but I wasn't able to test before I left town. I have a spare Gledopto remote I'll get setup when I get back.