hencou/esphome_components #11: Changing color with the colorwheel in HA

Ctrl-F4, 2022-05-03:
Very nice project!
Everything works great except when i choose a color with the colorwheel in HA. (bulb not changing colors, turns or stays white)

```
[22:08:58][D][light:035]: 'Test2' Setting:
[22:08:58][D][light:054]:   Color brightness: 8%
[22:08:58][D][light:057]:   Red: 25%, Green: 100%, Blue: 25%
[22:08:58][D][light:062]:   White: 7%
[22:08:58][D][light:084]:   Transition length: 0.2s
[22:08:58][D][mi.light:393]: Send Milight request: {"color_mode":"rgbw","state":"ON","brightness":71,"color":{"w":18},"white_value":18}
[22:08:58][D][mi.light:393]: Send Milight request: {"color_mode":"rgbw","state":"ON","brightness":71,"color":{"w":18},"white_value":18}
[22:08:58][D][mi.light:393]: Send Milight request: {"color_mode":"rgbw","state":"ON","brightness":71,"color":{"w":18},"white_value":18}
[22:08:58][D][mi.light:393]: Send Milight request: {"color_mode":"rgbw","state":"ON","brightness":71,"color":{"w":18},"white_value":18}
[22:08:58][D][mi.light:393]: Send Milight request: {"color_mode":"rgbw","state":"ON","brightness":71,"color":{"w":18},"white_value":18}
```

Maybe the Send Milight request syntax not complete?

hencou, 2022-05-04:
Hi,

Can you post your yaml configuration please? Maybe you are using "rgbw" mode? I have never tested that part because I am using only "rgb_cct". Currently color mode is determined by hue and saturation parts in the command, maybe rgbw is working on an other way.

hencou, 2022-05-04:
I think I have solved the issue in the meanwhile, can you test again? Please ensure ESPHome is downloading the newest version from Github instead of using the old cached code ;-)

Ctrl-F4, 2022-05-04:
Yes i use the "rgbw" mode. HA Colorwheel works now for 100% after this patch. Thank you very much! 

```
[19:29:20][D][light:035]: 'Test2' Setting:
[19:29:20][D][light:054]:   Color brightness: 55%
[19:29:20][D][light:057]:   Red: 53%, Green: 100%, Blue: 38%
[19:29:20][D][light:062]:   White: 48%
[19:29:20][D][light:084]:   Transition length: 0.2s
[19:29:20][D][mi.light:399]: Send Milight request: {"color_mode":"rgbw","state":"ON","brightness":23,"color":{"r":74,"g":140,"b":53,"w":122},"white_value":122}
[19:29:20][D][mi.light:399]: Send Milight request: {"color_mode":"rgbw","state":"ON","brightness":23,"color":{"r":74,"g":140,"b":53,"w":122},"white_value":122}
[19:29:20][D][mi.light:399]: Send Milight request: {"color_mode":"rgbw","state":"ON","brightness":23,"color":{"r":74,"g":140,"b":53,"w":122},"white_value":122}
[19:29:20][D][mi.light:399]: Send Milight request: {"color_mode":"rgbw","state":"ON","brightness":23,"color":{"r":74,"g":140,"b":53,"w":122},"white_value":122}
```
The "Color brightness" and "White brightness" in the HA colorwheel does not have a function for the rgbw type  i think? These sliders have no effect on the bulb, i don't mind.

hencou, 2022-05-04:
Hi,

The "Color brightness" and "White brightness" below the HA colorwheel have a function. While the "Brightness" slider above the HA colorwheel will control the "brightness" value in the request, the "Color brightness" will control all the "rgb" channels. The "White brightness" will controll the "w" channel and "white_value". I have tested it here, and I see the values in the json string changing as expected. At your side the values are not changing in the json result when sliding the brightness sliders?

Ctrl-F4, 2022-05-05:
Hi,

The rgbw bulbs never mix the rgb and white channel at the same time by design. 
Therefor the "Color brightness" and "White brightness" does not seem to have a funtion for this rgbw bulbs.
I checked with the https://github.com/sidoh/esp8266_milight_hub:

[img]https://i.imgur.com/fSlFyqH.png[/img]
[img]https://i.imgur.com/q5ERfu2.png[/img]

hencou, 2022-05-05:
Ok clear. It seems RGBW is missing a kind of "saturation" what is a relation between white and the RGB channels. HA shows that as 2 separate sliders White and Color brightness for RGBW then. I will close the case when there will be no solution for this then.

closed 2022-05-05 by hencou