hencou/esphome_components #14: lights flashing after upload

Ctrl-F4, 2022-09-04:
After uploading/install the firmware the lights are flashing 2 times. Is this expected behavior? Is it possible to prevent this behavior?

hencou, 2022-09-04:
There is no flashing behaviour in the firmware. Maybe you mean the lights are turning off, after a short time, after a (re)boot of the ESP module? This is expected behaviour, to prevent all lamps turning on after a power failure restore.

Ctrl-F4, 2022-09-04:
It is not a big issue for me as this happens only after upload/install the firmare (when bulb is off). Just noticed.

I just checked, power failure restore fails here.
When light bulb is on and esp module is rebooted by the reset button the button in HA switches briefly to off and to on again. (bulb stays on)
When light bulb is on and esp module is powercycled (unplug/plug) the button in HA switches to off and stays off. (bulb stays on!)

Perhaps difference in code for the "rgbw" and "rgb_cct"?
I only have these "rgbw" bulbs so can't compare.

hencou, 2022-09-04:
I have no "rgbw" to test with, but I should expect the same behaviour for all bulb types because this piece of code is executed before any bulb type is coming in. The only thing I can imagine is a transition issue, maybe you can try to add "default_transition_length: 0s" to the "light" part of the yaml file.

hencou, 2022-09-06:
Additional idea: when you configure the logging level of the ESPhome device to "DEBUG", maybe there is usefull info logged which will put you in the right direction

Ctrl-F4, 2022-09-09:
Hi, something i did not noticed before: when bulb is on and then the switch in HA is turned off, the bulb is turned off but after half a second flashes once. Seems same behaviour.  So did some debug with esphome flasher console while uploading OTA. 
This is the command esphome sends that flashes the bulb, directly after OTA:
`[D][mi:399]: Send Milight request: {"color_mode":"rgbw","state":"OFF","brightness":255,"color":{"r":255,"g":255,"b":255,"w":255},"white_value":255}`  
Very strange.  The "default_transition_length: 0s" made no difference.

hencou, 2022-09-12:
Hi, I changed the code a little so when turning Off a RGBW light, all other commands are removed from the JSON string command list. Can you test, and let me know of this solve your issue?

Ctrl-F4, 2022-09-18:
Hi, Yes issue is solved thanks again!

closed 2022-09-18 by Ctrl-F4