hencou/esphome_components #20: Mi - fut089

GertVanderstukken, 2023-11-07:
Hello,

I recently learned about this project and I really like the idea of using esphome for my Home Assistant / Mi integration.
Up until now I am using sidoh's esp8266_milight_hub. Also a great project BTW...

However, while testing this ESPhome integration, it seems that it is not working correctly with fut089 remote's.
I can send commands towards lights from Home Assistant without issues, however commands from the remote itself do not make it into Home Assistant (the lights itself are triggered correctly off course). 

Even more: when my ESPhome config only contains lights with fut089 as definition, debug mode does not show anything when pressing any button on the remote.
When I add a light with a remote_type: rgb_cct, that light is working correctly and the fut089 lights give something like this in debug mode when pressing on and off (on is making it into home assistant, off obviously not as it is detected wrong - it should be state: OFF).

As you can see: the packets are detected wrong: _**rgb_cct packet received (9 bytes)**_

```
[22:14:32][D][mi:067]: Received packet: 
rgb_cct packet received (9 bytes):
Raw packet: 84 54 73 A4 DA 45 79 DA 79 

Decoded:
Key      : 84
b1       : 25
ID       : 621D
Command  : 01
Argument : 01
Sequence : 5F
Group    : 01
Checksum : 32
[22:14:32][D][mi:128]: Received Milight request: {"state":"ON"}
[22:14:33][D][mi:067]: Received packet: 
rgb_cct packet received (9 bytes):
Raw packet: 40 98 3F D8 26 96 1A 26 5F 

Decoded:
Key      : 40
b1       : 25
ID       : 621D
Command  : 01
Argument : 0A
Sequence : 60
Group    : 01
Checksum : 88
[22:14:33][D][mi:128]: Received Milight request: {"command":"mode_speed_up"}
```

This is my (test) config:
```
substitutions:
  device_name: esphome-mi01
  friendly_name: esphome-mi01

esphome:
  name: $device_name
  friendly_name: $friendly_name

external_components:
  - source: github://hencou/esphome_components
    components: mi

esp8266:
  board: nodemcuv2

# Enable logging
logger:
  level: DEBUG

# Enable Home Assistant API
api:
  encryption:
    key: "<:)>"

ota:
  password: "<:)>"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Mi01 Fallback Hotspot"
    password: "<:)>"

captive_portal:

mi:
  id: mi1  #optional
  ce_pin: D2 #required, default: 4
  csn_pin: D8 #required, default: 15
  #reset_pin: 0 #optional, default: 0, only needed with LT8900 radio
  radio_interface_type: nrf24 #optional, default: nrf24, possible values: nrf24,lt8900
  packet_repeats: 50 #optional, default: 50, total of sended packets per command
  #listen_repeats: 3 #optional, default: 20, amount of received packets needed for a incoming command from other remote
  state_flush_interval: 5000 #optional, default: 10000, time in miliseconds to send the latest state report to HomeAssistant
  packet_repeat_throttle_threshold: 200 #optional, default: 200, threshold to limit the amount of packets in a second
  packet_repeat_throttle_sensitivity: 0 #optional, default: 0
  packet_repeat_minimum: 3 #optional, default: 3
  enable_automatic_mode_switching: false #optional, default: false
  rf24_power_level: MAX #optional, default: MAX, possible values: MIN, LOW, HIGH, MAX
  rf24_listen_channel: LOW #optional, default: LOW, possible values: LOW, MID, HIGH
  packet_repeats_per_loop: 10 #optional, default: 10, repeat packets sended per loop
  resend_last_command: true #optional, default: true, repeats the latest command after a random time between 2 and 3 seconds again
  rf24_channels: #optional, 1-3 values required when used, default: LOW-MID-HIGH, possible values: LOW, MID, HIGH
    - LOW
    - MID
    - HIGH
#  on_command_received: #optional, useful to send remote commands to HA and process them further there in automations
#    - homeassistant.event:
#        event: esphome.mi_command_received
#        data:
#          device_id: !lambda "return format_hex(data.device_id);"
#          group_id: !lambda "return data.group_id;"
#          remote_type: !lambda "return data.remote_type.c_str();"
#          command: !lambda "return data.command.c_str();"

light:
  - platform: mi
    id: light1
    name: testmi1
    device_id: 0x621D
    group_id: 1
    remote_type: fut089 
    default_transition_length: 0s

  - platform: mi
    id: light2
    name: testmi2
    device_id: 0x621D
    group_id: 2
    remote_type: fut089
    default_transition_length: 0s

  - platform: mi
    id: group1
    name: "Group1 test"
    device_id: 0x621D
    group_id: 0
    remote_type: fut089
    default_transition_length: 0s

  - platform: mi
    id: light3
    name: "testmi3"
    device_id: 0xC1A
    group_id: 1
    remote_type: rgb_cct
    default_transition_length: 0s

  - platform: mi
    id: light4
    name: "testmi4"
    device_id: 0xC1A
    group_id: 2
    remote_type: rgb_cct
    default_transition_length: 0s

  - platform: mi
    id: group2
    name: "Group2 test"
    device_id: 0xC1A
    group_id: 0
    remote_type: rgb_cct
    default_transition_length: 0s
```

GertVanderstukken, 2023-11-18:
Is anybody using other remote_type's then rgb_cct ?

ron-blom, 2023-11-20:
> Is anybody using other remote_type's then rgb_cct ?

I'm using both rgbw and rgb_cct.

hencou, 2023-11-20:
Only rgb_cct and fut091 here, so unfortunately not able to test and repoduce the problem here.

Bl00d-B0b, 2023-11-20:
Hello,

I have several milight hub's, so one can work in dump mode. another can be in esp home. and i can try to help you to reproduce a problem with fut091

Ed

Bl00d-B0b, 2023-11-21:
Looks like you can close this issue, and continue with #22

closed 2023-11-21 by hencou