hencou/esphome_components #21: Problems when compiling in update of ESPHOME

JanRunhart1, 2023-11-18:
Since the latest updates of ESPHome i have some trouble with updating and/or compiling the Milight component.

**_This is the errorcode:_**
`Compiling .pioenvs/esphome-web-56b5a8/src/esphome/components/mi/MiLightRadioConfig.cpp.o
src/esphome/components/mi/GroupAlias.cpp: In static member function 'static void GroupAlias::saveAliases(Stream&, const std::map<String, GroupAlias>&)':
src/esphome/components/mi/GroupAlias.cpp:56:17: error: call of overloaded 'write(int)' is ambiguous
   stream.write(0);
                 ^
In file included from /data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/Stream.h:26,
                 from src/esphome/components/mi/GroupAlias.h:1,
                 from src/esphome/components/mi/GroupAlias.cpp:1:
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/Print.h:61:20: note: candidate: 'virtual size_t Print::write(uint8_t)'
     virtual size_t write(uint8_t) = 0;
                    ^~~~~
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/Print.h:62:12: note: candidate: 'size_t Print::write(const char*)'
     size_t write(const char *str)
            ^~~~~
*** [.pioenvs/esphome-web-56b5a8/src/esphome/components/mi/GroupAlias.cpp.o] Error 1
========================= [FAILED] Took 20.18 seconds =========================
===== [ERROR] /config/esphome/esphome-web-56b5a8.yaml =====`

I've nothing changed in the yaml file and settings. Can anyone help me out with this?

hencou, 2023-11-18:
Hi Jan,

What happend when you try this example when you configure it as a new device:
https://github.com/hencou/esphome_components/blob/main/example_milight.yaml

JanRunhart1, 2023-11-19:
Hi @hencou,

I've got exactly the same problem when configuring a new device. I've checked the settings in the Yaml file.

**_Yaml file:_**
[yaml.txt.txt](https://github.com/hencou/esphome_components/files/13402878/yaml.txt.txt)

Errorlog:
`Compiling .pioenvs/esphome-web-56b5a8/src/esphome/components/mi/GroupStateCache.cpp.o
src/esphome/components/mi/GroupAlias.cpp: In static member function 'static void GroupAlias::saveAliases(Stream&, const std::map<String, GroupAlias>&)':
src/esphome/components/mi/GroupAlias.cpp:56:17: error: call of overloaded 'write(int)' is ambiguous
   stream.write(0);
                 ^
In file included from /data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/Stream.h:26,
                 from src/esphome/components/mi/GroupAlias.h:1,
                 from src/esphome/components/mi/GroupAlias.cpp:1:
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/Print.h:61:20: note: candidate: 'virtual size_t Print::write(uint8_t)'
     virtual size_t write(uint8_t) = 0;
                    ^~~~~
/data/cache/platformio/packages/framework-arduinoespressif32/cores/esp32/Print.h:62:12: note: candidate: 'size_t Print::write(const char*)'
     size_t write(const char *str)
            ^~~~~
*** [.pioenvs/esphome-web-56b5a8/src/esphome/components/mi/GroupAlias.cpp.o] Error 1
========================= [FAILED] Took 101.45 seconds =========================
`

tbrasser, 2023-11-19:
Same here:

Compiling .pioenvs/milighthub-one/src/esphome/components/mi/light/mi_light.cpp.o
src/esphome/components/mi/GroupAlias.cpp: In static member function 'static void GroupAlias::saveAliases(Stream&, const std::map<String, GroupAlias>&)':
src/esphome/components/mi/GroupAlias.cpp:56:17: error: call of overloaded 'write(int)' is ambiguous
   stream.write(0);
                 ^
In file included from /config/.esphome/platformio/packages/framework-arduinoespressif32/cores/esp32/Stream.h:26,
                 from src/esphome/components/mi/GroupAlias.h:1,
                 from src/esphome/components/mi/GroupAlias.cpp:1:
/config/.esphome/platformio/packages/framework-arduinoespressif32/cores/esp32/Print.h:61:20: note: candidate: 'virtual size_t Print::write(uint8_t)'
     virtual size_t write(uint8_t) = 0;
                    ^~~~~
/config/.esphome/platformio/packages/framework-arduinoespressif32/cores/esp32/Print.h:62:12: note: candidate: 'size_t Print::write(const char*)'
     size_t write(const char *str)
            ^~~~~
Compiling .pioenvs/milighthub-one/src/esphome/components/mi/mi.cpp.o
*** [.pioenvs/milighthub-one/src/esphome/components/mi/GroupAlias.cpp.o] Error 1

hencou, 2023-11-19:
It seems the problem was only when compling with ESP32, I am only using ESP8266. Seems also compiling now for ESP32 as well again (ensure you retrieve the new code from Github instead from cache)

JanRunhart1, 2023-11-19:
@hencou, thanks for the quick response. It's working now! I'm closing the issue.

closed 2023-11-19 by JanRunhart1