hencou/esphome_components #23: Group 0 does not remember brightness after on/off/on

Bl00d-B0b, 2023-11-20:
Hello,

when using Group 0 (all zones), when you turn it on, then setup brightness... turn it off and on again, brightness reset back to  0...

with other zones, 1, 2, 3 ,4 it's works as expected...

Ed

hencou, 2023-11-21:
Hi Ed,

Group 0 is a special thing. It is only used to send a command, and will have no status. When processing  a group 0 command, the status will be written to the underlying 1,2,3,4 zones. The reason there can be no status is because when one of the zones 1,2,3,4 changes, group 0 will not be in sync anymore.

Bl00d-B0b, 2023-11-21:
Hello. This one I already understand, how this logic works. but using remote control I can
1. ON zone 0
2. OFF zone 1 ,2
3. set Effect 5
4. OFF zone 0
5. ON zone 0
6. OFF zone 3, 4
7. set effect 6
8. OFF zone 0
9. ON zone 0

Then I have synchronized effect 5 on zone 1, 2 and another synchronized effect on zone 3, 4

If I follow this login:
1. ON zone 1
2. set effect 5
3. ON zone 2
4. set effect 5
5. on zone 3
6. set effect 6
7. ON zone 4
8. set effect 6

then effect 5 and effect 6 are not synchronized between zones

Ed

hencou, 2023-11-21:
HI Ed,

Its not clear for me, where do you see that "effect" state? Do you mean the "effect" state in HomeAssistant or somewhere else? Is only that "effect" state the problem or also something with the brightness?

Bl00d-B0b, 2023-11-21:
[image]

or "M" - Mode button on remote

hencou, 2023-11-21:
Hi Ed,

Unfortunaltely this is a restriction. The ESPHome light library has no option to set the effect when receiving it:
[image]

Bl00d-B0b, 2023-11-22:
Hello,

the screenshot I send you in previous mail is from your code. It's supported, and I can chose Effects :)
the problem is not with effect, but with logic. when zone 0 is used. in Remote control zone 0 is the same, like zone 1-4.

Ed

hencou, 2023-11-22:
HI Ed,

What I tried to mention is that an effect command (sent by a remote and received by an ESPhome module) will never be sent to HomeAssistant as with other values as on/off, brightness etc. 

So HomeAssistant will not reflect an effect command from a remote command.

closed 2023-11-24 by hencou