hencou/esphome_components #24: Any Ideas how to bridge or Sync ID's

Bl00d-B0b, 2023-11-21:
Hello,

I have rgb_cct Wall switch ID:0x02CB and rgb_cct Remote Control ID:0x1F83. Both paired and manages same groups of Mi LED's. 

Is it possible to configure like a "bridge" button, which can see ON/OFF if any of  my Switch or remote control ON/OFF light???

Maybe another implementation? Like Syncing buttons... Just try to think... Any Ideas?

Ed

Bl00d-B0b, 2023-11-23:
hello,

thanks, i'll try

Bl00d-B0b, 2023-11-23:
Is it possible to implement within same esphome, without homeassistant. just in .yaml file.

hencou, 2023-11-23:
I don't know how, maybe there is a ESPhome component enabling communication between them? Otherwise an MQTT broker between them can maybe an option.

hencou, 2023-11-25:
Closed: implementing a "bridge" is more ESPhome or HomeAssistant related.

closed 2023-11-25 by hencou