hencou/esphome_components #25: Compilation failure with stripped down yaml

szupi-ipuzs, 2023-11-22:
Hi, I am trying to use your great mi component, but I only wanted the part with "on_command_received", so I stripped down the example yaml to this:

```yaml
substitutions:
  device_name: mi-rf-remote-handler
  friendly_name: Mi RF Remote Handler

esphome:
  name: $device_name
  # platformio_options:                     #optional, to set additional build flags
  #   build_flags:                          #optional, to set additional build flags
  #     - "-D USE_ESP32_ALTERNATE_SPI"      #optional, to use the alternate HSPI SPI bus instead of the default VSPI on an ESP32 board
  #     - "-D ALT_SPI_MISO_PIN=15"          #optional, if HSPI bus is used, alternate pins can also be defined
  #     - "-D ALT_SPI_MOSI_PIN=14"
  #     - "-D ALT_SPI_SCLK_PIN=12"
  #     - "-D ALT_SPI_SS_PIN=4"

external_components:
  - source:
      type: local
      path: /components/hencou_esphome_components/components
    components: mi

esp8266:
  board: nodemcuv2

# Enable logging
logger:
  level: WARN

captive_portal:

mdns:

api:
  password: ""

ota:
  password: ""

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  fast_connect: true
  ap:
    ssid: $device_name
    password: !secret fallback_wifi_password

mi:
  id: mi1  #optional
  ce_pin: D2 #required, default: 4
  csn_pin: D8 #required, default: 15
  reset_pin: 0 #optional, default: 0, only needed with LT8900 radio
  radio_interface_type: nrf24 #optional, default: nrf24, possible values: nrf24,lt8900
  packet_repeats: 50 #optional, default: 50, total of sended packets per command
  listen_repeats: 20 #optional, default: 20, amount of received packets needed for a incoming command from other remote
  state_flush_interval: 5000 #optional, default: 10000, time in miliseconds to send the latest state report to HomeAssistant
  packet_repeat_throttle_threshold: 200 #optional, default: 200, threshold to limit the amount of packets in a second
  packet_repeat_throttle_sensitivity: 0 #optional, default: 0
  packet_repeat_minimum: 3 #optional, default: 3
  enable_automatic_mode_switching: false #optional, default: false
  rf24_power_level: MAX #optional, default: MAX, possible values: MIN, LOW, HIGH, MAX
  rf24_listen_channel: LOW #optional, default: LOW, possible values: LOW, MID, HIGH
  packet_repeats_per_loop: 10 #optional, default: 10, repeat packets sended per loop
  resend_last_command: true #optional, default: true, repeats the latest command after a random time between 2 and 3 seconds again
  rf24_channels: #optional, 1-3 values required when used, default: LOW-MID-HIGH, possible values: LOW, MID, HIGH
    - LOW
    - MID
    - HIGH
  on_command_received: #optional, useful to send remote commands to HA and process them further there in automations
    - homeassistant.event:
        event: esphome.mi_command_received
        data:
          device_id: !lambda "return format_hex(data.device_id);"
          group_id: !lambda "return data.group_id;"
          remote_type: !lambda "return data.remote_type.c_str();"
          command: !lambda "return data.command.c_str();"
```

Unfortunately this does not compile then, because:
```
In file included from src/esphome/components/mi/mi.cpp:1:
src/esphome/components/mi/mi.h:5:10: fatal error: esphome/components/light/light_state.h: No such file or directory
    5 | #include "esphome/components/light/light_state.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
```

It seems the LightState package is not added automatically in such case, yet you still use it. I was trying to add an import line in __init__.py, but it still wasn't compiling...
Eventually I used the whole example yaml and this time it worked.

hencou, 2023-11-22:
Yes you need to include a "light" section in your yaml. The module will receive commands configured from the "light" section then. so with the same id and groupId and remote_type.

szupi-ipuzs, 2023-11-22:
The point is I don't have any compatible lights, I only own a remote and want to use it to control my other lights. So it seemed kind of strange that I still need to add a light in yaml... But ok, I understand now that this is by design.
Thanks!

closed 2023-11-22 by szupi-ipuzs

hencou, 2023-11-22:
You can "capture" the ID and remote type of your remote: when you use this example:
https://github.com/hencou/esphome_components/blob/main/example_milight.yaml
..and set the logger level on "DEBUG": then you will see the received commands in the logging, with the ID and remote type of your remote. Configure a light with this values and you are ready to go.