hencou/esphome_components #26: Support for RGBCCT Miboxer S2-W

szupi-ipuzs, 2023-11-22:
Hi, I have[ this nice little remote RGBCCT](https://miboxer.com/pl/produkt/2-4g-tecza-remote-rgbcct) and it seems it almost works with your mi component (rgb_cct remote type).

Out of 4 physical buttons I get response on 3 of them (see the pictures):
- _ON_ - _seems to match_, I get:
```
[17:26:54][D][mi:067]: Received packet: 
rgb_cct packet received (9 bytes):
Raw packet: 1C BC F1 F5 02 6D CC 02 37 

Decoded:
Key      : 1C
b1       : 25
ID       : 94DA
Command  : 01
Argument : 01
Sequence : 9A
Group    : 01
Checksum : 84
[17:26:54][VV][api.service:184]: send_homeassistant_service_response: HomeassistantServiceResponse {
  service: 'esphome.mi_command_received'
  data: HomeassistantServiceMap {
  key: 'device_id'
  value: '94da'
}
  data: HomeassistantServiceMap {
  key: 'group_id'
  value: '1'
}
  data: HomeassistantServiceMap {
  key: 'remote_type'
  value: 'rgb_cct'
}
  data: HomeassistantServiceMap {
  key: 'command'
  value: '{"state":"ON"}'
}
  is_event: YES
}
```
- _OFF_ - not a match (**different command**), but keypress is decoded:
```
[17:28:38][D][mi:067]: Received packet: 
rgb_cct packet received (9 bytes):
Raw packet: F4 E4 59 CD 6A CA 25 6A AD 

Decoded:
Key      : F4
b1       : 25
ID       : 94DA
Command  : 01
Argument : 0A
Sequence : 9B
Group    : 01
Checksum : F6
[17:28:38][VV][api.service:184]: send_homeassistant_service_response: HomeassistantServiceResponse {
  service: 'esphome.mi_command_received'
  data: HomeassistantServiceMap {
  key: 'device_id'
  value: '94da'
}
  data: HomeassistantServiceMap {
  key: 'group_id'
  value: '1'
}
  data: HomeassistantServiceMap {
  key: 'remote_type'
  value: 'rgb_cct'
}
  data: HomeassistantServiceMap {
  key: 'command'
  value: '{"command":"mode_speed_up"}'
}
  is_event: YES
}
```
- _Color icon button_ - not a match (**different command**), but keypress is decoded. Also the group id is different then with the rest:
```
[17:33:13][D][mi:067]: Received packet: 
rgb_cct packet received (9 bytes):
Raw packet: C1 F1 2C 8C F9 C5 42 FA DB 

Decoded:
Key      : C1
b1       : 25
ID       : 94DA
Command  : 01
Argument : 14
Sequence : 9D
Group    : 01
Checksum : 3F
[17:33:13][VV][api.service:184]: send_homeassistant_service_response: HomeassistantServiceResponse {
  service: 'esphome.mi_command_received'
  data: HomeassistantServiceMap {
  key: 'device_id'
  value: '94da'
}
  data: HomeassistantServiceMap {
  key: 'group_id'
  value: '15'
}
  data: HomeassistantServiceMap {
  key: 'remote_type'
  value: 'rgb_cct'
}
  data: HomeassistantServiceMap {
  key: 'command'
  value: '{"state":"OFF"}'
}
  is_event: YES
}
```
- _Brightness(?) button_ - **no reaction**, no keypress decoded  (there is even no "Received packet" log).

There is also a touch wheel that seems to be correctly decoded, giving value "mode" in range 0-100:
```
[17:36:20][D][mi:067]: Received packet: 
rgb_cct packet received (9 bytes):
Raw packet: 24 B4 09 FD FE 36 D8 FA F4 

Decoded:
Key      : 24
b1       : 25
ID       : 94DA
Command  : 05
Argument : 56
Sequence : 9E
Group    : 01
Checksum : D9
[17:36:20][VV][api.service:184]: send_homeassistant_service_response: HomeassistantServiceResponse {
  service: 'esphome.mi_command_received'
  data: HomeassistantServiceMap {
  key: 'device_id'
  value: '94da'
}
  data: HomeassistantServiceMap {
  key: 'group_id'
  value: '1'
}
  data: HomeassistantServiceMap {
  key: 'remote_type'
  value: 'rgb_cct'
}
  data: HomeassistantServiceMap {
  key: 'command'
  value: '{"mode":86}'
}
  is_event: YES
}
```

I will try using all possible remote types and see if any of them gives better result.

hencou, 2023-11-22:
Hi,

Nice remote, I was not aware of it. Seems most compatible with "fut089".

GertVanderstukken, 2023-11-23:
I think this is related to #20 and #22.
When you press OFF, it say': value: '{"command":"mode_speed_up"}'

This is because the packet's are interpreted as rgb_cct packets.
In your case, I guess you actually configured rgb_cct as device_type?

 can you try to set the device_type to fut089? 
However: I'm afraid that you will get the same result...

szupi-ipuzs, 2023-11-23:
@GertVanderstukken , yes, seems to be the same problem, I do have it setup as rgb_cct. Will later try all of them and see which matches the best. However, I am worried a little about this problem:

> Brightness(?) button - no reaction, no keypress decoded (there is even no "Received packet" log).

@hencou Any way to debug this?

szupi-ipuzs, 2023-11-23:
No, fut089 doesn't seem to work at all for me. However I get a bit better result with fut091 - this time pressing "off" correctly causes
```
value: '{"state":"OFF"}'
```
although the decoded group id is 5... However pressing "color", also results in
```
value: '{"state":"OFF"}'
```
but with group id 15...

Pressing "brightness" does not trigger any reaction again.

Here's my yaml:
```yaml
substitutions:
  device_name: mi-rf-remote-handler
  friendly_name: Mi RF Remote Handler

esphome:
  name: $device_name
  # platformio_options:                     #optional, to set additional build flags
  #   build_flags:                          #optional, to set additional build flags
  #     - "-D USE_ESP32_ALTERNATE_SPI"      #optional, to use the alternate HSPI SPI bus instead of the default VSPI on an ESP32 board
  #     - "-D ALT_SPI_MISO_PIN=15"          #optional, if HSPI bus is used, alternate pins can also be defined
  #     - "-D ALT_SPI_MOSI_PIN=14"
  #     - "-D ALT_SPI_SCLK_PIN=12"
  #     - "-D ALT_SPI_SS_PIN=4"

external_components:
  - source: github://hencou/esphome_components
    components: mi

esp8266:
  board: nodemcuv2

# Enable logging
logger:
  level: VERY_VERBOSE

captive_portal:

mdns:

api:
  password: ""

ota:
  password: ""

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password
  ap:
    ssid: $device_name
    password: !secret fallback_wifi_password

mi:
  id: mi1  #optional
  ce_pin: D2 #required, default: 4
  csn_pin: D8 #required, default: 15
  reset_pin: 0 #optional, default: 0, only needed with LT8900 radio
  radio_interface_type: nrf24 #optional, default: nrf24, possible values: nrf24,lt8900
  packet_repeats: 50 #optional, default: 50, total of sended packets per command
  listen_repeats: 20 #optional, default: 20, amount of received packets needed for a incoming command from other remote
  state_flush_interval: 5000 #optional, default: 10000, time in miliseconds to send the latest state report to HomeAssistant
  packet_repeat_throttle_threshold: 200 #optional, default: 200, threshold to limit the amount of packets in a second
  packet_repeat_throttle_sensitivity: 0 #optional, default: 0
  packet_repeat_minimum: 3 #optional, default: 3
  enable_automatic_mode_switching: false #optional, default: false
  rf24_power_level: MAX #optional, default: MAX, possible values: MIN, LOW, HIGH, MAX
  rf24_listen_channel: HIGH #optional, default: LOW, possible values: LOW, MID, HIGH
  packet_repeats_per_loop: 10 #optional, default: 10, repeat packets sended per loop
  resend_last_command: true #optional, default: true, repeats the latest command after a random time between 2 and 3 seconds again
  rf24_channels: #optional, 1-3 values required when used, default: LOW-MID-HIGH, possible values: LOW, MID, HIGH
    - LOW
    - MID
    - HIGH
  on_command_received: #optional, useful to send remote commands to HA and process them further there in automations
    - homeassistant.event:
        event: esphome.mi_command_received
        data:
          device_id: !lambda "return format_hex(data.device_id);"
          group_id: !lambda "return data.group_id;"
          remote_type: !lambda "return data.remote_type.c_str();"
          command: !lambda "return data.command.c_str();"

light:
  - platform: mi #required
    id: light1 #required
    name: ${friendly_name} #required
    device_id: 0xAB01 #required, hexadacimal value of MiLight id
    group_id: 1 #required, 1-4 or 1-8, depending on remote type
    remote_type: fut091 #required, possible values: rgb_cct, rgb, cct, rgbw, fut089, fut091, fut020
    default_transition_length: 0s #optional, but 0s gives a better behaviour instead the default 200ms
  # Set these to calibrate the color temperature of your light, measured with an external color temp. sensor or app
  # optional, [153, 370] mireds is the range miboxer uses internally ([6535, 2702] K)
    #cold_white_color_temperature: 6500 K
    #warm_white_color_temperature: 2700 K
# optional variables: all variables of ESPHome base light component
```

hencou, 2023-11-24:
Lets proceed om #22

closed 2023-11-24 by hencou