hencou/esphome_components #28: Expose only selected functions of light to HA

TroLoos, 2023-12-14:
Hi, I just realized that there is an ESPHome version of MH - great work!

I already started to think about migrating, but I am looking for one specific option that I use in my current setup... can you check if it's possible to achieve with this code?

I have 2 types of lights @ home - RGBW and only White LED strips. I control all of them with RGB+CCT physical and MH remotes. When I expose these devices to HA, for White lights I only choose to expose on/off and brightness change functions, so when this device is passed further to HomeKit, I only have option to change brightness, and not color. This is how such device looks like in HA:

[image]

And here's the code of such light, with some parts disabled (only choose changing brightness and light effects):

light:
    - name: "Bright @ Dinner Table [L]"
      command_topic: milight/0x67B4/fut089/1
      state_topic: milight/states/0x67B4/fut089/1
      schema: json
      qos: 1
      optimistic: false
      retain: true
      **color_temp: false
      rgb: false
      brightness: true
      effect: true
      effect_list: [night_mode]**
      availability_topic: milight/client_status
      payload_available: 'connected'
      payload_not_available: 'disconnected'

Is there any way to pick and choose functions like that in your project? It is great when you pass these devices to HomeKit - on my phone I don't have a color wheel for lights that are only white. And I retain compatibility with my physical remotes that I use to pair / unpair and control lights.

hencou, 2023-12-14:
When you configure renote_type "cct" or "fut091" in stead of "rgb_cct" you should only be able to change only the light temperature and brightness. THe available functions id depending on the chooses remote_type, the possible values are: rgb_cct, rgb, cct, rgbw, fut089, fut091, fut020.

Example of how "fut091" looks in HA:
[image]

TroLoos, 2023-12-15:
Hi, yes - I understand that. But it's not what I'm looking for... here's why:

I have physical remotes paired to my MiLight controllers - and I would like to have MiLight devices that are synced to those remotes so when I change the light using such remote, it will be reflected on MiLight Hub and also in Home Assistant entity. I understand that I can pair another types of remotes to controller, but it will be out of sync if I use a physical one. So the only way is to mimic ID of a physical remote and it will sync automatically to HA.

That's why in my setup right now I specify on MQTT integration level which functions of light should be enabled and which not. From your example - I would also want to disable ability to change light temperature which is not the case in my LED controller and strip. I'm sure this kind of light definition can be done on ESP level.

hencou, 2023-12-15:
Maybe its possible but will take a lot of effort to combine this in conjunction of the remote_types variable. Depending on this configured variable, one or more options should be available to configure as an option. Then the default options should be taken from the configured remote, and being overwritten when an additional variable like "temperature" on/off is configured.

TroLoos, 2023-12-15:
Yeah, I got it. Right now I am looking for a solution to create template light on top of auto-populated light from esphome MH, that will limit capabilities of different light entities. Since I am no expert in template sensors, no success yet but I think it is a direction that should work. Thanks a lot for taking a look into my case.

closed 2023-12-21 by hencou

theyo-tester, 2024-01-16:
Hi @TroLoos 
mybe this is a little off-topic, but this should be possible, is no big issue. For instance, I combined in a light template from another entity (knx for ON/OFF), and the rest of the functions, I took from the mi-light hub entity from the esphome integration. 

The knx part is not relevant for you, but I let it there for reference, in case someone finds this.

```
light:
   - platform: template
     lights:
       bad_licht_comb:
         friendly_name: "Bad Licht"
         value_template: "{{ states('light.licht_bad') }}"
         turn_on:
           service: light.turn_on
           entity_id: light.licht_bad, light.milightgateway_bad    # I send the comand to both entities, knx and milight
         turn_off:
           service: light.turn_off
           entity_id: light.licht_bad, light.milightgateway_bad
         level_template: "{{ state_attr('light.milightgateway_bad', 'brightness')|d(0,true)|int }}"  # this is for status update
         temperature_template: "{{ state_attr('light.milightgateway_bad', 'color_temp') }}"  # you may NOT specify this line, to avoid white temperature setting
         set_level:   # this would set and send the actual brightness, to the hub
           service: light.turn_on
           entity_id: light.milightgateway_bad
           data:
             brightness: "{{ brightness }}"
         set_temperature:  # do NOT set this to avoid the white color setting
           service: light.turn_on
           entity_id: light.milightgateway_bad
           data_template:
             color_temp: "{{ color_temp }}"
         min_mireds_template: "{{ state_attr('light.milightgateway_bad', 'min_mireds') }}"   # don't set this to avoid the white color setting
         max_mireds_template: "{{ state_attr('light.milightgateway_bad', 'max_mireds') }}"  # don't set this to avoid the white color setting