hencou/esphome_components #29: Twitchy turning ON\OFF CCT Lamps

VladNord, 2023-12-20:
Hello! This problem has already been described previously, but I cant find it in issues.
CCT laps turn on and off somewhat twitchy. RGBW works as should. 
Here is the logs from console with turning on\off process of cct and rgbw.
I don’t understand where the commands to dim the brightness come from when turning the lamp on and off

```
#CCT
00:06:05	[D]	[light:047]	State: ON
00:06:05	[D]	[mi:128]	Received Milight request: {"state":"ON"}
00:06:06	[D]	[mi:128]	Received Milight request: {"command":"brightness_up"}
00:06:06	[D]	[mi:128]	Received Milight request: {"command":"brightness_up"}
00:06:07	[D]	[mi:128]	Received Milight request: {"command":"brightness_up"}
00:06:07	[D]	[mi:128]	Received Milight request: {"command":"brightness_up"}
00:06:07	[D]	[mi:128]	Received Milight request: {"command":"brightness_up"}
00:06:07	[D]	[mi:128]	Received Milight request: {"command":"brightness_up"}
00:06:07	[D]	[mi:128]	Received Milight request: {"command":"brightness_up"}
00:06:08	[D]	[mi:128]	Received Milight request: {"command":"brightness_up"}
00:06:08	[D]	[mi:128]	Received Milight request: {"command":"brightness_up"}
00:06:08	[D]	[mi:128]	Received Milight request: {"command":"brightness_up"}

00:06:16	[D]	[light:036]	'Koridor_L4' Setting:
00:06:16	[D]	[light:047]	  State: OFF
00:06:17	[D]	[mi:128]	Received Milight request: {"command":"brightness_down"}
00:06:17	[D]	[mi:128]	Received Milight request: {"command":"brightness_down"}
00:06:17	[D]	[mi:128]	Received Milight request: {"command":"brightness_down"}
00:06:17	[D]	[mi:128]	Received Milight request: {"command":"brightness_down"}
00:06:17	[D]	[mi:128]	Received Milight request: {"command":"brightness_down"}
00:06:18	[D]	[mi:128]	Received Milight request: {"command":"brightness_down"}
00:06:18	[D]	[mi:128]	Received Milight request: {"command":"brightness_down"}
00:06:18	[D]	[mi:128]	Received Milight request: {"command":"brightness_down"}
00:06:18	[D]	[mi:128]	Received Milight request: {"command":"brightness_down"}
00:06:19	[D]	[mi:128]	Received Milight request: {"command":"brightness_down"}
00:06:19	[D]	[mi:128]	Received Milight request: {"state":"OFF"}	

#RGBW

00:07:36	[D]	[light:047]	  State: ON
00:07:36	[D]	[mi:128]	Received Milight request: {"state":"ON"}
00:07:36	[D]	[mi:128]	Received Milight request: {"command":"set_white"}
00:07:37	[D]	[mi:128]	Received Milight request: {"brightness":255}

00:08:49	[D]	[light:036]	'Koridor_L8' Setting:
00:08:49	[D]	[light:047]	  State: OFF
00:08:49	[D]	[mi:128]	Received Milight request: {"brightness":0}
00:08:49	[D]	[mi:128]	Received Milight request: {"state":"OFF"}
```

Here is part of config for mi of esphome. WEB-server is enable (as far as I remember, a year ago the presence of this parameter caused a compilation error)

```
esphome:

   platformio_options:                     #optional, to set additional build flags
     build_flags:                          #optional, to set additional build flags
       - "-D USE_ESP32_ALTERNATE_SPI"      #optional, to use the alternate HSPI SPI bus instead of the default VSPI on an ESP32 board
       - "-D ALT_SPI_MISO_PIN=12"          #optional, if HSPI bus is used, alternate pins can also be defined
       - "-D ALT_SPI_MOSI_PIN=13"
       - "-D ALT_SPI_SCLK_PIN=14"
       - "-D ALT_SPI_SS_PIN=5"

external_components:
  - source: github://hencou/esphome_components
    components: mi

mi:
  
  id: mi1
  ce_pin: 4
  csn_pin: 5
  reset_pin: 0
  radio_interface_type: nrf24
  packet_repeats: 50
  listen_repeats: 20
  state_flush_interval: 5000
  packet_repeat_throttle_threshold: 200
  packet_repeat_throttle_sensitivity: 0
  packet_repeat_minimum: 3
  enable_automatic_mode_switching: false
  rf24_power_level: MAX
  rf24_listen_channel: LOW
  packet_repeats_per_loop: 10
  resend_last_command: false
  rf24_channels: #optional, 1-3 values required when used, default: LOW-MID-HIGH, possible values: LOW, MID, HIGH
    - LOW
    - MID
    - HIGH

light:
  - platform: mi
    id: Koridor_L4
    name: Koridor_L4
    device_id: 0x4
    group_id: 1
    remote_type: cct
    default_transition_length: 0s

  - platform: mi
    id: Koridor_L8
    name: Koridor_L8
    device_id: 0x8
    group_id: 2
    remote_type: rgbw
    default_transition_length: 0s
```

Thank you in advance

hencou, 2023-12-21:
Hi,

The ESPhome component sets the brightness to the minimum level, before switching the lights off. This is to prevent strange behavoir when switching the lights later on, and they can be on a maximum brightness, and then go to the wanted brightness.

The problem with the "cct" lights is: they have no method to set the brighness directly to a certain level. This is done in multiple steps, till the wanted value is reached.

In my own setup I solved the issue by changing the "remote_type" from "cct" to "fut091": this remote is able to set the brigness directly to a certain value, and solved the problem.

VladNord, 2023-12-21:
> In my own setup I solved the issue by changing the "remote_type" from "cct" to "fut091": this remote is able to set the brigness directly to a certain value, and solved the problem.

thanks for fast reply.
tried to change cct to fut091 - lamps don't react at all. Also tried fut020, fut089- same.

hencou, 2023-12-21:
I didn't remember exactly how I did -- maybe you need to pair the lamps again with the fut091 type.

VladNord, 2024-01-05:
Re-pairing lapmps with fut091 solved this Problem.
Weird, but with sidoh mi-light hub everything worked as it should with CCT remote.

Anyway, thanks for your work!

closed 2024-01-05 by VladNord