hencou/esphome_components #38: Group 0 really doesn't remember brightness after on/off/on

lilphil, 2025-04-12:
This issue is related https://github.com/hencou/esphome_components/issues/23 but as the issue seems to talk more about effects I thought it was best to open a new one.

I've just migrated from sidoh/esp8266_milight_hub, however I almost exclusively control the lights using group 0 on each device (largely because I like to have each light individually addressable, so every light gets a group 1-4!), and immediately found that the brightness is not restored after an off.

I've spent a bit of time today looking at the code and I've found the offending area which causes it to break for me. There are two blocks in MiLightClient which set the brightness to zero when the state is OFF:

```
  //<added by HC>
  if (status == MiLightStatus::OFF) {
    this->updateBrightness(0);
  }
  //</added by HC>
```

I realise you were saying that group 0 cannot understand the resulting state when the underlying groups differ, however the bulbs themselves remember the state. Additionally, it does seem that the brightness state is sent with an "on" anyway.

I guess you were trying to fix a different issue with this? But I can say that for group zero usage this breaks. I've removed these locally and it works well.

hencou, 2025-04-13:
Hi lilphil,

From what I remember it's mainly because I use this ESP modules behind a push wallswitch. Therefore, when I switch off the lights. they need to be on the lowest brightness when click on a wallswitch the next time. A long button press will result in increasing brightness.

Should be possible to make this behaviour (Brightness = 0 at Off command) configurable by a YAML setting in ESPHome.

lilphil, 2025-04-13:
I wonder if that would be achievable using an on_state or on_turn_off actions as part of the light platform to send a "set brightness to zero on off", or the on_command_received configuration instead of having another configuration for it?

closed 2025-11-22 by hencou